Seeed-Studio/ArduinoCore-imxrt #2: Install / Arduino IDE integration

vectorselector, 2020-06-03:
Hello,
I'm very excited to try this with the Archmix.
I dont' see the Archmix nor mxrt as options when I add the Seeed-Studio board list to the Arduino IDE.
https://raw.githubusercontent.com/Seeed-Studio/Seeed_Platform/master/package_seeeduino_boards_index.json
Which one is the Archmix?

Also, I am unclear about what cmake is supposed to build, and how this gets plugged into the Arduino IDE... also where/how the SDK is found through Arduino, etc.

Thank you

LynnL4, 2020-06-04:
Hi,
please use this url .
```http://files.seeedstudio.com/arduino/package_seeeduino_local_boards_index.json```

vectorselector, 2020-06-04:
Excellent, thank you LynnL4!
I now see the board in the list.

(I am on Raspberry Pi and it says "tool bossac not available" but I will research this, as it appears to be an Atmel flash loader tool and Archmix imxrt is an NXP chip not Atmel. Do you know anything about platform limitations for host?)

thank you

LynnL4, 2020-06-04:
Hi, 
Currently, only three platforms are supported. I will update the firmware of raspberry pie to you in the near future. please refer [here](https://github.com/Seeed-Studio/BOSSA)

vectorselector, 2020-06-15:
Hello, So, I tried more usual Linux 64bit intel on Ubuntu and Arduino board manager can't find the files when i try to install from the link above, which DOES list the Archmix imxrt board.
`Error downloading http://192.168.5.153/files.seeedstudio.com/arduino/core/imxrt/ArduinoCore-imxrt-1.1.0.tar.gz`

LynnL4, 2020-06-16:
http://files.seeedstudio.com/arduino/package_seeeduino_boards_index.json

closed 2020-06-16 by LynnL4