Seeed-Studio/ArduinoCore-imxrt #4: run code on board?

vectorselector, 2020-08-26:
I've tried with both Arduino and PlatformIO but cannot seem to run code on the board.

Arduino builds project OK but all project examples (even Archmix examples like SPI and Wire) return the following:
"An error occurred while uploading the sketch. Device unsupported"

PlatformIO when attempting merely "Run without Debugging f5" 
Error: Currently, PlatformIO does not support debugging for `Seeed Arch Mix`.
Please request support at https://github.com/platformio/platformio-core/issues 
or visit -> https://docs.platformio.org/page/plus/debugging.html

additionally, there is no basic example for PlatformIO usage with Archmix, so unable to confirm if working.

LynnL4, 2020-08-27:
Hi, 
   you need to upload bootloader at first, [bootloader.hex](https://github.com/Seeed-Studio/ArduinoCore-imxrt/blob/master/bootloaders/arch_mix/bootloader.hex). 
   How to upload bootloader please refer [here](https://github.com/JayHeng/NXP-MCUBootUtility).

vectorselector, 2020-09-01:
I have not been able to successfully use this bootloader tool with the Archmix board yet.  I'm on Linux,  it's not showing Wine regedit mapped serial UART ports, and the tool author refuses to port a python tool to Linux, citing factually incorrect statistics with a rather shaky sequence of (il)logical statements.
https://github.com/JayHeng/NXP-MCUBootUtility/issues/115

NXP offers this "Secure Provisioning" tool now:
https://www.nxp.com/design/software/development-software/mcuxpresso-software-and-tools-/mcuxpresso-secure-provisioning-tool:MCUXPRESSO-SECURE-PROVISIONING?&tab=Design_Tools_Tab

It's not working on my system due to Python problems... 

[29932] Error loading Python lib '/opt/nxp/MCUX_Provi_v2/bin/libpython3.7m.so.1.0': dlopen: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.25' not found (required by /opt/nxp/MCUX_Provi_v2/bin/libpython3.7m.so.1.0)

That's from a packaged NXP-released .deb file that supposedly is a standalone app...

Pillar1989, 2020-09-01:
Does "Secure Provisioning" Tool work? Or use a download tool like Jlink.

vectorselector, 2020-09-01:
The "secure provisioning" tool did not work,  python problems despite the tool being a packaged released .deb file for debian/ubuntu linux... 
i don't have a segger jlink device. 
thank you for your help, 

it's possible that windows users can follow this plan to upload the requisite firmware to the archmix to allow usage with the Arduino IDE  and the Platform IO IDE.

I would say that this need for special firmware must be on the front page of this repo inside the instructions.
https://github.com/Seeed-Studio/ArduinoCore-imxrt

This need to upload special firmware and the required tools was not mentioned in the Readme.md

I can close this issue, if you want, or leave it open.
I won't be able to spend any more time on this board.
I appreciate your help, the issues appear larger than your individual kind willingness to help: 

I've not been able to use it with any IDE not even the MCUxpresso.
It's not the same as the NXP RT1052 Eval board which is the closest supported board directly from NXP. 
If Seeed studio could provide out-of-the-box support for this board under ANY IDE that would allow users to program the board and begin working with it. 
I appreciate Mr Styger's project files, but this was not sufficient to get me blinking any LED via MCUxpresso.
Copy-pasting micro-python into a serial terminal REPL is not going to suffice for anything more than HelloWorld. Anything beyond this appears currently unsupported or experimental.

LynnL4, 2020-09-01:
Sorry about that, when did you buy arch mix? The new version of Arch Mix should come with this firmware by default, the old version does not by default.

closed 2020-11-12 by Pillar1989