Seeed-Studio/ArduinoCore-imxrt #5: How to upload the program to the Arch Mix board?

loboris, 2020-11-11:
Have anyone successfully **uploaded any program** to the board?

I have an **Seeed Arch Mix board** preloaded with RT-Thread, and it works with loaded firmware.

I've followed the instructions in this repository and I can successfully build a program both in Arduino and PlatformIO, but **uploading** the program simply **does not work**!

It was mentioned in https://github.com/Seeed-Studio/ArduinoCore-imxrt/issues/4 that maybe I should upload the **bootloader** first, but **how exactly do I do that** ?

The suggested [NXP-MCUBootUtility](https://github.com/JayHeng/NXP-MCUBootUtility) does not work, reporting that it cannot put MCU in  SDL (Serial Downloader Programmer) mode, **how do I use it** ?
The same thing is with [MCUXpresso Secure Provisioning Tool](https://www.nxp.com/design/software/development-software/mcuxpresso-software-and-tools-/mcuxpresso-secure-provisioning-tool:MCUXPRESSO-SECURE-PROVISIONING?&tab=Design_Tools_Tab).
What is the point in providing the [bootloader.hex](https://github.com/Seeed-Studio/ArduinoCore-imxrt/blob/master/bootloaders/arch_mix/bootloader.hex) without giving any instructions how to use it.

**Please, provide the complete instructions how to program the _Arch Mix board_, otherwise this board is completely useles.**

@LynnL4
I supose you have tested this repository on an acctual board (including the provided example), so it must be working.
**Please provide more information.**

Pillar1989, 2020-11-12:
@loboris  Maybe you can use JLink to burn this bootloader

loboris, 2020-11-12:
I only have LPC-Link2 (which can also run Segger J-Link firmware), but I was not able to flash anything to ArchMix with it (it comunicates with ArchMix via SWD, but reports the MCU cannot be stopped or something like that).

It looks like the RT-Thread bootloder on ArchMix boar actually works, I could upload the the prebuilt firmware (found in [Tools](https://files.seeedstudio.com/wiki/GM6020/res/Firmware_and_Tools.zip)), using instructions found in [Arch Mix Firmware Guide](https://files.seeedstudio.com/wiki/Arch_Mix/res/micropython_firmware.pdf).

In Adruino "_Device unsupported_" is reported when trying to flash. So either a different bootloader is needed, Arduino is not using **RTT Win USB** (ArchMix RT-Thread bootloader USB port), or something else is wrong...

loboris, 2020-11-12:
In **PlatformIO**, "_Device unsupported_" is also reported:

```
Processing seeed_arch_mix (platform: nxpimxrt; board: seeed_arch_mix; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxpimxrt/seeed_arch_mix.html
PLATFORM: NXP i.MX RT (3.8.0) > Seeed Arch Mix
HARDWARE: MIMXRT1052XXXXB 600MHz, 512KB RAM, 7.39MB Flash
PACKAGES:
 - framework-arduinoimxrt 1.2.1
 - tool-bossac 1.9.1-seeeduino
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 2 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Checking size .pio\build\seeed_arch_mix\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   0.9% (used 4728 bytes from 524288 bytes)
Flash: [          ]   0.3% (used 23640 bytes from 7747072 bytes)
Configuring upload protocol...   
AVAILABLE: sam-ba
CURRENT: upload_protocol = sam-ba
Looking for upload port...       
Auto-detected: COM5
Forcing reset using 1200bps open/close on port COM5
Waiting for the new upload port...
Uploading .pio\build\seeed_arch_mix\firmware.bin

Device unsupported
*** [upload] Error 1
=============================================================== [FAILED] Took 7.96 seconds ===============================================================
```

LynnL4, 2020-11-13:
Hi
FYI https://wiki.seeedstudio.com/Arch_Mix/#flashing-arduino-bootloader-to-arch-mix

loboris, 2020-11-13:
Thank you @LynnL4 

I couldn't use the **J-Link** method because I have **LPC-Link2** (with Segger J-Link firmware) and it is **not supported** by **J-Flash**.
(BTW, the cheapest J-Link working with J-Flash costs ~500 US$ !)

The method with **NXP Boot Utility** worked, and I've successfully uploaded the **Arduino bootloader**.

It is works fine, both in
**Arduino**:
```
Skica koristi 22948 bytes (0%) od prostora za program. Maximum je  8323072 bajtova.
Device       : IMRXT1052
Version      : v1.0 [Arduino:Z] Oct  9 2020 15:16:54
Address      : 0x60010000
Pages        : 32768
Page Size    : 256 bytes
Total Size   : 8192KB
Planes       : 1
Lock Regions : 32
Locked       : none
Security     : false
Boot Flash   : false
BOD          : false
BOR          : false
Write 32628 bytes to flash (128 pages)
[==============================] 100% (128/128 pages)
Done in 4.714 seconds
Verify 32628 bytes of flash
[==============================] 100% (128/128 pages)
Verify successful
Done in 0.149 seconds
```
and in **PlatformIO**:
```
> Executing task: C:\Users\Korisnik\.platformio\penv\Scripts\platformio.exe run --target upload <

Processing seeed_arch_mix (platform: nxpimxrt; board: seeed_arch_mix; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxpimxrt/seeed_arch_mix.html
PLATFORM: NXP i.MX RT (3.8.0) > Seeed Arch Mix
HARDWARE: MIMXRT1052XXXXB 600MHz, 512KB RAM, 7.39MB Flash
PACKAGES:
 - framework-arduinoimxrt 1.2.1
 - tool-bossac 1.9.1-seeeduino
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 2 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Checking size .pio\build\seeed_arch_mix\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   0.9% (used 4728 bytes from 524288 bytes)
Flash: [          ]   0.3% (used 23640 bytes from 7747072 bytes)
Configuring upload protocol...
AVAILABLE: sam-ba
CURRENT: upload_protocol = sam-ba
Looking for upload port...
Auto-detected: COM6
Forcing reset using 1200bps open/close on port COM6
Waiting for the new upload port...
Uploading .pio\build\seeed_arch_mix\firmware.bin
Write 32876 bytes to flash (129 pages)

[                              ] 0% (0/129 pages)
[=                             ] 3% (5/129 pages)
...
[==============================] 100% (129/129 pages)
Done in 4.444 seconds
Verify 32876 bytes of flash

[=                             ] 3% (5/129 pages)
...
[==============================] 100% (129/129 pages)
Verify successful
Done in 0.206 seconds
============================================================== [SUCCESS] Took 15.59 seconds ==============================================================
Terminal will be reused by tasks, press any key to close it.
```

Thanks again, just wondering why this information wasn't available before (ArchMix board apeared some 2 years ago and I've read the wiki couple of times and I've never saw this part), but **better late than never**.

I would suggest to put this information in this repository's README.md

LynnL4, 2020-11-13:
Thanks for your feedback, we will put this information in this repository's README.md ASAP.

closed 2020-11-13 by LynnL4