Seeed-Studio/ArduinoCore-imxrt #6: Using MCUXpresso Secure Provisioning to Flash Arduino Bootloader

loboris, 2020-11-14:
How to use **NXP-MCUBootUtility** to flash th Arduino bootloader was explained [here](https://wiki.seeedstudio.com/Arch_Mix/#using-nxpbootutility-to-flash-arduino-bootloader).
Unfortunatelly, this tool **only supports Windows**, here is a solution to this issue.

(To the repository maintainer: this is not exactly an issue, but I'm not shure where else I could post it, maybe it could be added to the ArchMix Wiki,  feel free to suggest...)

<hr>

There is also another tool, the [**MCUXpresso Secure Provisioning Tool**](https://www.nxp.com/design/software/development-software/mcuxpresso-software-and-tools-/mcuxpresso-secure-provisioning-tool:MCUXPRESSO-SECURE-PROVISIONING) which can also be used under **Linux**.

However, there is a small problem with the Seeed's provided Arduino **bootloader.hex** file.
As stated in the Secure Provisioning Tool documentation:
> For current version of MCUXpresso Provisioning Tools, build image for images with XIP boot header is not
> supported. Remove XIP boot header of the image before building the image, as XIP boot header is currently added
> to the image by MCUXpresso Provisioning Tools.

_bootloader.hex_ contains the **XIP boot header** (the first 8KB) which must be **removed** from it.
I've prepared the modified bootloader (with removed XIP header) and flashing it with **MCUXpresso Secure Provisioning Tool** now works.
Download [ArchMix_Arduino_bootloader.hex](https://loboris.eu/ArchMix/ArchMix_Arduino_bootloader.hex) and use it instead of the original hex file.

How to use the **MCUXpresso Secure Provisioning Tool**:

**1.** Download the _MCUXpresso Secure Provisioning Tool_ for your operating system, you will need to register at NXP site. You can also download the Linux package from [here](https://loboris.eu/ArchMix/MCUXPRESSO_SECURE_PROV_V2_SIGNED_AMD64.deb.bin))
**2.** Install the tool (you may need to make the downloaded file executable first)
**3.** After the tool is installed start it (you can find it in the in the menu under _Programming_)
**4.** Make sure the **flax-spi-nor/IS25LPxxxA_IS25WPxxxA** is selected under _Boot Device_.
**5.** For _Source executable image_ select the downloaded _ArchMix_Arduino_bootloader.hex_ file
[image]

**6.** Click on **Build image** button to build flashable binnary
[image]

**7.** Put your ArcMix board into **_ROM Bootloader mode_** by connecting **Rx to +3.3V** and setting the **BOOT_CFG** switches to **0001** (with USB cable unplugged)
**8.** Plug in the USB cable and click on **USB** button to check if the board was recognized
[image]

**8.** Switch to **Write image** tab and click on **Write image** button to write the bootloader image.
It should succeed with the following message:
[image]

**9.** Unplug the USB cable, **disconnect Rx from +3V** and set the **BOOT_CFG** switches to **0010**

<hr>

_You can now use the board with **Arduino** and/or **PlatformIO** (both works also under Linux)._

[image]

eugenio, 2020-12-03:
Do you know if this bootloader and the method to flash it into the flash-spi-rom would work also with other microconterollers of the same family such as:
MIMXRT1062DLV6A
MIMXRT1062DVJ6A
and more importantly on the lower end member of the family IMXRT10xx ?

loboris, 2020-12-05:
I think it should work.
It should also work for any hex file which has a XIP header if you remove it, MCUXpresso Secure Provisioning Tool adds its own header,

closed 2020-12-08 by Pillar1989