Seeed-Studio/ArduinoCore-imxrt #9: Pin Definition for Arch Mix User Switch?

CarmonColvin, 2021-07-07:
I have the Arduino bootloader on an Arch Mix board.
I am having difficulty getting any software connection to the "SW2 USER" switch on the board.

The Arch_Mix.pdf document states this.
> SW2 User Button For user configure, for this development board No. 125 pin is SW2

The example python code has this as an example.
`key = Pin(("KEY", 125), Pin.IN, Pin.PULL_UP) #Set pin 125 to pull-up input mode`

Based on that information, I assumed this Arduino equivalent would do the same.
`pinMode(125, INPUT_PULLUP);`
It does not. This will hard crash the board; requiring a power cycle to regain control. I thought I bricked it the first time I tried this.

What is the proper pin number for the user switch?

Can that be added as a definition to /variants/arch_mix/variant.h the same way the USER_LED_R, USER_LED_G, USER_LED_B pins are defined?

Thanks.

Pillar1989, 2021-07-07:
@CarmonColvin  Wellcome PR of this.

closed 2021-07-07 by Pillar1989

CarmonColvin, 2021-07-07:
What is the proper pin number for the user switch on the Arch Mix board running Arduino?